mapmot/arduino-overlay #2: cross-avr/binutils - "ld" for avr cannot find linker script "avr5.x"

mapmot, 2017-10-13:
Read https://bugs.gentoo.org/147155 for more details and lots of hilarious comments towards the end. The linking workaround proposed there works.

sarnold, 2017-12-10:
The latest binutils-config has ldscripts symlink handling.  All the boards in the menu "should" work; I've testing building several but I can only upload to Uno and Mega (for avr).

mapmot, 2017-12-18:
Thank you for the update. Do you happen to know the first version of binutils-config that works? Then we can make it a ${DEPEND}.

sarnold, 2017-12-18:
Whatever the latest version is. binutils-config-5-r3 I think?

mapmot, 2017-12-19:
The cleanest solution in my opinion is the following patch in /etc/portage/patches/cross-avr/binutils
--- ld/Makefile.in
+++ ld/Makefile.in
@@ -386,7 +386,7 @@
 # We put the scripts in the directory $(scriptdir)/ldscripts.
 # We can't put the scripts in $(datadir) because the SEARCH_DIR
 # directives need to be different for native and cross linkers.
-scriptdir = $(tooldir)/lib
+scriptdir = $(libdir)
 BASEDIR = $(srcdir)/..
 BFDDIR = $(BASEDIR)/bfd
 INCDIR = $(BASEDIR)/include

mapmot, 2017-12-19:
The cleanest solution in my opinion is the following patch in /etc/portage/patches/cross-avr/binutils
--- ld/Makefile.in
+++ ld/Makefile.in
@@ -386,7 +386,7 @@
-scriptdir = $(tooldir)/lib
+scriptdir = $(libdir)
 BASEDIR = $(srcdir)/..
 BFDDIR = $(BASEDIR)/bfd
 INCDIR = $(BASEDIR)/include

mapmot, 2018-02-12:
Still doesn't work for me with sys-devel/binutils-config-5.1-r1. The patch above is still the only thing that works.