mapmot/arduino-overlay #3: /usr/libexec/gcc/arm-none-eabi/ld: cannot find -lc_nano

mapmot, 2017-10-13:
This needs cross-arm-none-eabi/newlib-nano. An attempt can be found in sys-libs/newlib[nano]. This was taken from https://forums.gentoo.org/viewtopic-t-1017098-start-0.html, but I am unable to get it to work.

sarnold, 2017-12-19:
Did you try "--specs=nano.specs"?

sarnold, 2017-12-19:
Here's what I have installed on the udoo board, it compiled and flashed demo sketches just fine using 5.4.0 and a slightly modified version of the USE="nano" ebuild on the forum post:

```
user@udoo ~ $ find /usr/ -mount -name \*nano.so\*
user@udoo ~ $ find /usr/ -mount -name \*nano.a\*
/usr/arm-none-eabi/lib/librdimon_nano.a
/usr/arm-none-eabi/lib/armv6-m/librdimon_nano.a
/usr/arm-none-eabi/lib/armv6-m/libc_nano.a
/usr/arm-none-eabi/lib/armv6-m/libg_nano.a
/usr/arm-none-eabi/lib/fpu/librdimon_nano.a
/usr/arm-none-eabi/lib/fpu/libc_nano.a
/usr/arm-none-eabi/lib/fpu/libg_nano.a
/usr/arm-none-eabi/lib/libc_nano.a
/usr/arm-none-eabi/lib/armv8-m.base/librdimon_nano.a
/usr/arm-none-eabi/lib/armv8-m.base/libc_nano.a
/usr/arm-none-eabi/lib/armv8-m.base/libg_nano.a
/usr/arm-none-eabi/lib/armv7e-m/librdimon_nano.a
/usr/arm-none-eabi/lib/armv7e-m/softfp/librdimon_nano.a
/usr/arm-none-eabi/lib/armv7e-m/softfp/fpv5-sp-d16/librdimon_nano.a
/usr/arm-none-eabi/lib/armv7e-m/softfp/fpv5-sp-d16/libc_nano.a
/usr/arm-none-eabi/lib/armv7e-m/softfp/fpv5-sp-d16/libg_nano.a
/usr/arm-none-eabi/lib/armv7e-m/softfp/libc_nano.a
/usr/arm-none-eabi/lib/armv7e-m/softfp/libg_nano.a
/usr/arm-none-eabi/lib/armv7e-m/softfp/fpv5-d16/librdimon_nano.a
/usr/arm-none-eabi/lib/armv7e-m/softfp/fpv5-d16/libc_nano.a
/usr/arm-none-eabi/lib/armv7e-m/softfp/fpv5-d16/libg_nano.a
/usr/arm-none-eabi/lib/armv7e-m/fpu/librdimon_nano.a
/usr/arm-none-eabi/lib/armv7e-m/fpu/fpv5-sp-d16/librdimon_nano.a
/usr/arm-none-eabi/lib/armv7e-m/fpu/fpv5-sp-d16/libc_nano.a
/usr/arm-none-eabi/lib/armv7e-m/fpu/fpv5-sp-d16/libg_nano.a
/usr/arm-none-eabi/lib/armv7e-m/fpu/libc_nano.a
/usr/arm-none-eabi/lib/armv7e-m/fpu/libg_nano.a
/usr/arm-none-eabi/lib/armv7e-m/fpu/fpv5-d16/librdimon_nano.a
/usr/arm-none-eabi/lib/armv7e-m/fpu/fpv5-d16/libc_nano.a
/usr/arm-none-eabi/lib/armv7e-m/fpu/fpv5-d16/libg_nano.a
/usr/arm-none-eabi/lib/armv7e-m/libc_nano.a
/usr/arm-none-eabi/lib/armv7e-m/libg_nano.a
/usr/arm-none-eabi/lib/libg_nano.a
/usr/arm-none-eabi/lib/armv7-m/librdimon_nano.a
/usr/arm-none-eabi/lib/armv7-m/libc_nano.a
/usr/arm-none-eabi/lib/armv7-m/libg_nano.a
/usr/arm-none-eabi/lib/thumb/librdimon_nano.a
/usr/arm-none-eabi/lib/thumb/libc_nano.a
/usr/arm-none-eabi/lib/thumb/libg_nano.a
/usr/arm-none-eabi/lib/armv7-ar/thumb/librdimon_nano.a
/usr/arm-none-eabi/lib/armv7-ar/thumb/softfp/librdimon_nano.a
/usr/arm-none-eabi/lib/armv7-ar/thumb/softfp/libc_nano.a
/usr/arm-none-eabi/lib/armv7-ar/thumb/softfp/libg_nano.a
/usr/arm-none-eabi/lib/armv7-ar/thumb/fpu/librdimon_nano.a
/usr/arm-none-eabi/lib/armv7-ar/thumb/fpu/libc_nano.a
/usr/arm-none-eabi/lib/armv7-ar/thumb/fpu/libg_nano.a
/usr/arm-none-eabi/lib/armv7-ar/thumb/libc_nano.a
/usr/arm-none-eabi/lib/armv7-ar/thumb/libg_nano.a
/usr/arm-none-eabi/lib/armv8-m.main/librdimon_nano.a
/usr/arm-none-eabi/lib/armv8-m.main/softfp/fpv5-sp-d16/librdimon_nano.a
/usr/arm-none-eabi/lib/armv8-m.main/softfp/fpv5-sp-d16/libc_nano.a
/usr/arm-none-eabi/lib/armv8-m.main/softfp/fpv5-sp-d16/libg_nano.a
/usr/arm-none-eabi/lib/armv8-m.main/softfp/fpv5-d16/librdimon_nano.a
/usr/arm-none-eabi/lib/armv8-m.main/softfp/fpv5-d16/libc_nano.a
/usr/arm-none-eabi/lib/armv8-m.main/softfp/fpv5-d16/libg_nano.a
/usr/arm-none-eabi/lib/armv8-m.main/fpu/fpv5-sp-d16/librdimon_nano.a
/usr/arm-none-eabi/lib/armv8-m.main/fpu/fpv5-sp-d16/libc_nano.a
/usr/arm-none-eabi/lib/armv8-m.main/fpu/fpv5-sp-d16/libg_nano.a
/usr/arm-none-eabi/lib/armv8-m.main/fpu/fpv5-d16/librdimon_nano.a
/usr/arm-none-eabi/lib/armv8-m.main/fpu/fpv5-d16/libc_nano.a
/usr/arm-none-eabi/lib/armv8-m.main/fpu/fpv5-d16/libg_nano.a
/usr/arm-none-eabi/lib/armv8-m.main/libc_nano.a
/usr/arm-none-eabi/lib/armv8-m.main/libg_nano.a
```

mapmot, 2017-12-19:
It works now using the ebuild in the overlay. I honestly don't remember what the problem was.

closed 2017-12-19 by mapmot