mapmot/arduino-overlay #11: Roadmap

mapmot, 2017-12-18:
A very interesting read:

https://blog.arduino.cc/2017/10/11/be-among-the-first-to-try-arduino-ide-1-9-beta

"The other is arduino-preprocessor, which supersedes ctags in the sketch preprocessing phase. Moving to a different tool has been a necessary step for many reasons, the most important being the ctags’ limited parsing of complex C++ sketches."

"AVR core has been moved to its own repo."

I guess the logical step would be to split the package into:

dev-embedded/arduino: Just the java IDE
dev-embedded/arduino-hardware with use flags xxx,yyy,zzz
dev-embedded/arduino-core-xxx
dev-embedded/arduino-core-yyy
dev-embedded/arduino-core-zzz

modelled after gstreamer, gst-plugins-meta and gst-plugin-xxx

sarnold, 2017-12-19:
The package split seems fine, but I think the "proper" way to build avr/cortex-M toolchains is the multilib way (although there are lots of patches in the upstream GNU ARM toolchain.  Building 7.2.0 arm-none-eabi from ada overlay now (no extra arm embedded patches for now, those are mainly on 6.4.0).

mapmot, 2017-12-19:
I wonder what the *proper* gentoo way is when dealing  with crossdev and ebuilds - for instance we have a sys-libs/newlib ebuild that should only be called via cross-arm-none-eabi/newlib - I have created the link in the overlay, but it seems unelegant. Also the binutils patch in #2, that should only be used for cross-avr/binutils - is there an overlay equivalent for /etc/portage/patches?

mapmot, 2017-12-19:
The proper way to build avr/cortex-M *is* multilib. This needs fine tuning the --with-multilib-list= gcc option in crossdev. Unfortunately this is extremely poorly documented other that the --with-multilib-list=rmprofile that everybody is using and I gave up on trying to grasp it after a couple of spectacular failures and went the no-multilib way. Also, since I'm doing everything natively on arm, space is at premium....

mapmot, 2017-12-19:
Just out of mild curiosity:
cat /var/db/pkg/cross-arm-none-eabi/gcc-7.2.0/SIZE
120326621

What does your full blown multilib say?

My last attempt at a multilib cross-arm-none-eabi took two and a half days on an arm host. If all a user wants to do is blink some leds, it will probably be way beyond his/her attention span even on amd64...

sarnold, 2017-12-19:
$ cat /var/db/pkg/cross-arm-none-eabi/gcc-7.2.0/SIZE 
285838142

Aaaand it fails miserably on the upstream (arduino) sam platform tree for due, even worse than 6.4.0 does.  In this case the failing linker scripts are part of the platform trees and when i tried the upstream linker scripts (the sam3e8x-something equivalent to arduino due) they didn't do any better.  The avr stuff is a little different as far as linker args (and simpler) and works fine with all gcc up to 7.2.0 so what was your solution for your sam device?  (meaning flags and linker script)

sarnold, 2017-12-19:
Note that crossdev arm-none-eabi 5.4.0 built with multilib-list=aprofile in EXTRA_ECONF works smooth as butter with the existing platform flags/linker scripts.  It just pukes with anything newer.

mapmot, 2017-12-20:
Due is SAM(Cortex-M3), of which I have none :) Zero is SAMD(Cortex-M0+), of which I have several, and they all work with 7.2.0 in the Arduino environment with the default platform.txt modified for the location of the toolchain and CMSIS.
The linker scripts and args are all located within the CMSIS tree, so you could try downloading upstream CMSIS from https://github.com/ARM-software/CMSIS_5 and shoving a Device subfolder from https://github.com/arduino/ArduinoModule-CMSIS-Atmel
In my experience, the mere presence of hardfloat tuples in a multilib toolchain causes it to fail spectacularly. We need a custom multilib-list=gentoo-embedded ...
You could also try letting the Board Manager download a board package for you and checking the multilib list...

mjeveritt, 2017-12-20:
I'm game for some testing .. got a Mega1280/ Various Uno-clones, an M0-Pro, Nano, Mini etc.

My dev PC is a bit ancient, so I'm working with gcc-4.9 chains presently, but I can probably fix up a 6.4 crossdev on request.

mapmot, 2017-12-20:
@mjeveritt, everything AVR should work out-of-the-box with this overlay (Mega1280/ Various Uno-clones, Nano, Mini etc.)
You will probably need > gcc-5 toolchain for that, since Arduino takes advantage of lto optimizations now.
For the M0-Pro you can just let the Arduino board manager download a board package for you, it should work without crossdev toolchain installed.