mapmot/arduino-overlay #12: patch -p1  failed with /var/tmp/portage/dev-embedded/arduino-1.8.5/files/arduino-fix-utoa-to-match-name.patch

bdouxx, 2017-12-19:
* ERROR: dev-embedded/arduino-1.8.5::arduino failed (prepare phase):
 *   patch -p1  failed with /var/tmp/portage/dev-embedded/arduino-1.8.5/files/arduino-fix-utoa-to-match-name.patch
 * 
 * Call stack:
 *               ebuild.sh, line  124:  Called src_prepare
 *             environment, line 4491:  Called default
 *      phase-functions.sh, line  853:  Called default_src_prepare
 *      phase-functions.sh, line  918:  Called __eapi6_src_prepare
 *             environment, line  326:  Called eapply '/var/tmp/portage/dev-embedded/arduino-1.8.5/files/arduino-1.8.5-startup.patch' '/var/tmp/portage/dev-embedded/arduino-1.8.5/files/arduino-1.8.5-platform.patch' '/var/tmp/portage/dev-embedded/arduino-1.8.5/files/arduino-1.8.5-remove-avr-gcc-tools-dependency.patch' '/var/tmp/portage/dev-embedded/arduino-1.8.5/files/arduino-fix-utoa-to-match-name.patch' '/var/tmp/portage/dev-embedded/arduino-1.8.5/files/arduino-sam-platform-toolchain.patch'
 *             environment, line  775:  Called _eapply_patch '/var/tmp/portage/dev-embedded/arduino-1.8.5/files/arduino-fix-utoa-to-match-name.patch'
 *             environment, line  713:  Called __helpers_die 'patch -p1  failed with /var/tmp/portage/dev-embedded/arduino-1.8.5/files/arduino-fix-utoa-to-match-name.patch'
 *   isolated-functions.sh, line  117:  Called die
 * The specific snippet of code:
 *              die "$@"
 * 
 * If you need support, post the output of `emerge --info '=dev-embedded/arduino-1.8.5::arduino'`,
 * the complete build log and the output of `emerge -pqv '=dev-embedded/arduino-1.8.5::arduino'`.
!!! When you file a bug report, please include the following information:
GENTOO_VM=icedtea-bin-8  CLASSPATH="" JAVA_HOME="/opt/icedtea-bin-3.6.0"
JAVACFLAGS="-source 1.8 -target 1.8" COMPILER=""
and of course, the output of emerge --info =arduino-1.8.5
 * The complete build log is located at '/var/log/portage/dev-embedded:arduino-1.8.5:20171219-155707.log'.
 * For convenience, a symlink to the build log is located at '/var/tmp/portage/dev-embedded/arduino-1.8.5/temp/build.log'.
 * The ebuild environment file is located at '/var/tmp/portage/dev-embedded/arduino-1.8.5/temp/environment'.
 * Working directory: '/var/tmp/portage/dev-embedded/arduino-1.8.5/work/Arduino-1.8.5'
 * S: '/var/tmp/portage/dev-embedded/arduino-1.8.5/work/Arduino-1.8.5'

[dev-embedded:arduino-1.8.5:20171219-155707.log](https://github.com/mapmot/arduino-overlay/files/1572297/dev-embedded.arduino-1.8.5.20171219-155707.log)

mapmot, 2017-12-19:
@bdouxx thanks for the feedback.
This is caused by #10 
@sarnold should look into this.
In the meanwhite, I have reverted that merge. You should checkout the repository again, or resync if using repoman and retry.

sarnold, 2017-12-19:
sorry about that, those patches should have been conditional on udoo use flag (and now they are).

closed 2020-12-04 by bdouxx