mapmot/arduino-overlay #15: arduino blink sketch compil but nothing happen

bdouxx, 2017-12-21:
Hi,

I have bought this card:
https://fr.aliexpress.com/item/Pro-Mini-ATmega328P-Board-Pro-Mini-UNO-Shield-Adapter-CH340G-USB-to-TTL-Programmer-easy-to/32782075100.html

device is recognized:
> [ 6977.077291] usb 1-2: new full-speed USB device number 7 using xhci_hcd
> [ 6977.258806] usb 1-2: New USB device found, idVendor=1a86, idProduct=7523
> [ 6977.258808] usb 1-2: New USB device strings: Mfr=0, Product=2, SerialNumber=0
> [ 6977.258809] usb 1-2: Product: USB2.0-Serial
> [ 6977.259481] ch341 1-2:1.0: ch341-uart converter detected
> [ 6977.259910] usb 1-2: ch341-uart converter now attached to ttyUSB0

here the programm:
[Blink.txt](https://github.com/mapmot/arduino-overlay/files/1579861/Blink.txt)

On my lenovo T460 (dualboot with windows 10),

It's ok for windows, code compile and card blink:
[arduino_windows.txt](https://github.com/mapmot/arduino-overlay/files/1579859/arduino_windows.txt)
but nothing blink with linux:
[arduino_linux.txt](https://github.com/mapmot/arduino-overlay/files/1579858/arduino_linux.txt)

It's my first test with arduino's world, so I have no idea where is the problem.

can you help me?

mapmot, 2017-12-21:
Did you install ctags from the overlay? Any other ctags installed in your system? What is the output of eselect ctags list?

bdouxx, 2017-12-21:
thanks a lot

It was version 20161028 which doesn't work but with version 99999999 from this overlay it's ok.

closed 2020-12-04 by bdouxx