mapmot/arduino-overlay #17: Download issue stops build

globalcitizen, 2018-01-27:
```
untar-unzip-download-web:
      [get] Getting: https://github.com/arduino-libraries/WiFi101-FirmwareUpdater-Plugin/releases/download/v0.9.1/WiFi101-Updater-ArduinoIDE-Plugin-0.9.1.zip
      [get] To: /var/tmp/portage/dev-embedded/arduino-1.8.5/work/Arduino-1.8.5/build/shared/WiFi101-Updater-ArduinoIDE-Plugin-0.9.1.zip
      [get] https://github.com/arduino-libraries/WiFi101-FirmwareUpdater-Plugin/releases/download/v0.9.1/WiFi101-Updater-ArduinoIDE-Plugin-0.9.1.zip moved to https://github-production-release-asset-2e65be.s3.amazonaws.com/63938396/7470e9f2-1dd8-11e7-8088-b4f38eb33d28?X-Amz-Algorithm=AWS4-HMAC-SHA256&X-Amz-Credential=AKIAIWNJYAX4CSVEH53A%2F20180127%2Fus-east-1%2Fs3%2Faws4_request&X-Amz-Date=20180127T044319Z&X-Amz-Expires=300&X-Amz-Signature=feb1ff2499545aa4e18876a1af053891ba6c6f0c31b61f252918aa733146b017&X-Amz-SignedHeaders=host&actor_id=0&response-content-disposition=attachment%3B%20filename%3DWiFi101-Updater-ArduinoIDE-Plugin-0.9.1.zip&response-content-type=application%2Foctet-stream
      [get] Error getting https://github.com/arduino-libraries/WiFi101-FirmwareUpdater-Plugin/releases/download/v0.9.1/WiFi101-Updater-ArduinoIDE-Plugin-0.9.1.zip to /var/tmp/portage/dev-embedded/arduino-1.8.5/work/Arduino-1.8.5/build/shared/WiFi101-Updater-ArduinoIDE-Plugin-0.9.1.zip

untar-unzip-download-local:

untar-unzip-checksum:
     [echo] Testing checksum of "shared/WiFi101-Updater-ArduinoIDE-Plugin-0.9.1.zip"
 [checksum] Could not find file /var/tmp/portage/dev-embedded/arduino-1.8.5/work/Arduino-1.8.5/build/shared/WiFi101-Updater-ArduinoIDE-Plugin-0.9.1.zip to generate checksum for.

BUILD FAILED
/var/tmp/portage/dev-embedded/arduino-1.8.5/work/Arduino-1.8.5/build/build.xml:105: The following error occurred while executing this line:
/var/tmp/portage/dev-embedded/arduino-1.8.5/work/Arduino-1.8.5/build/build.xml:598: The following error occurred while executing this line:
/var/tmp/portage/dev-embedded/arduino-1.8.5/work/Arduino-1.8.5/build/build.xml:191: The following error occurred while executing this line:
/var/tmp/portage/dev-embedded/arduino-1.8.5/work/Arduino-1.8.5/build/build.xml:767: Could not find file /var/tmp/portage/dev-embedded/arduino-1.8.5/work/Arduino-1.8.5/build/shared/WiFi101-Updater-ArduinoIDE-Plugin-0.9.1.zip to generate checksum for.
```

globalcitizen, 2018-01-27:
I feel like altering the ebuild to download these files first would be a good way to improve it.

Benefits:
- Parallelization / pre-downloading while deps are building (portage)
- Proper checksum support (portage)
- Failed builds do not need to re-download deps

mapmot, 2018-01-27:
pull requests are welcome ;)

closed 2018-01-27 by mapmot