mapmot/arduino-overlay #18: Bossa digest verification failed - size

manugutito, 2018-01-31:
Hi,

I just configured and synced this overlay at my gentoo machine. It syncs successfully, but when I try to emerge arduino-1.8.5, I get the following error:

```
!!! Digest verification failed:
!!! /var/lib/arduino-overlay/dev-embedded/bossa/bossa-2017.06.16.ebuild
!!! Reason: Filesize does not match recorded size
!!! Got: 667
!!! Expected: 653
```

I've re-synced it a couple of times, but nothing changes.

I don't know if this is the right place to post this, so feel free to point me elsewhere.

Thanks!

mapmot, 2018-01-31:
Try cloning again, should be fixed now.

manugutito, 2018-01-31:
Now I get this:

'''
/usr/bin/git pull
fatal: refusing to merge unrelated histories
!!! git pull error in /var/lib/arduino-overlay
q: Updating ebuild cache in /var/lib/arduino-overlay ... 
q: Finished 25 entries in 0.000198 seconds

Action: sync for repo: arduino-overlay, returned code = 128
'''

This is the first time I try to install something outside of portage, so I'm sorry if this is an obvious thing.

globalcitizen, 2018-02-01:
Came to report same.

Not fixed with current repo contents.

```
Calculating dependencies - * Digest verification failed:
 * /var/lib/arduino-overlay/dev-embedded/bossa/bossa-2017.06.16.ebuild
 * Reason: Filesize does not match recorded size
 * Got: 667
 * Expected: 653
... done!
```

globalcitizen, 2018-02-01:
```
# cd /var/lib/arduino-overlay/dev-embedded/bossa
# ebuild bossa-2017.06.16.ebuild manifest
>>> Creating Manifest for /var/lib/arduino-overlay/dev-embedded/bossa
# git diff Manifest
-EBUILD bossa-2017.06.16.ebuild 653 SHA256 b9daf89da6a4f45a40301d52fc143e84ff3c567008d69478ba2299b83d45193a SHA512 8d902b63f57a2979836c34db681cbe88fd0ee2d88d965eb9cddeee434a3f0b784fce21dc97d81a59a7f8cb5f9c16e288e9a503c5e8780333a102f2ae9c1ba4a5 WHIRLPOOL c450234dbcdde52aef322294814dfb0509bf2470344839d6504595e44455035dc66bed1befc1e1ef2588cef436851a32a5cf7961f126f1a23014d755fd70a8b7
+EBUILD bossa-2017.06.16.ebuild 667 SHA256 9d4eebf395aa22c12d4cc4d644f72695bb979b231aaef2c4c0a753524fe63d8d SHA512 9b13283825c9a82a6c952597efb4414dbb8e6d4815e93dce3e5c7181a4cbde5099a05b62568fb2ea5f4baae151b8d7403676a1dae9b5223cee4968dd9ba4b74a WHIRLPOOL b3b1a55596cbbdf3dcc0b7a4dbf2602cf1d1acc7ec9adfe4230278ac97ff35f7a3aff1e6c1012828c6beb97fe903bda214fc4b1a2de08fb157076e352d49bed7
```

mapmot, 2018-02-01:
@manugutito should do 'sudo rm -rf /var/lib/arduino-overlay' and 'emerge --sync'

This shoud be fixed with the latest commit [https://github.com/mapmot/arduino-overlay/commit/d7bd70aa1a293a29685ea0f8a38605b014472ee1](url)

manugutito, 2018-02-01:
Thanks! Now I can sync and I don't get the error while trying to emerge arduino-1.8.5. However, I don't see bossa among the packages to be emerged (so of course no digest verification failed), and I get an error while emerging listserialportsc-1.0.7. Is it possible that this issue is related?

The error I get is:
```
!!! ERROR: The active vm could not be found
```
and I think this is relevant:
```
 * QA Notice: file does not exist:
 * 
 *      dolib: liblistSerialsj.so.1.0.7 does not exist
!!! When you file a bug report, please include the following information:
GENTOO_VM=  CLASSPATH="" JAVA_HOME=""
JAVACFLAGS="" COMPILER=""
```

mapmot, 2018-02-02:
@manugutito , you are running into #5