mapmot/arduino-overlay #20: arduino-builder cannot find dependencies

carbofos, 2018-02-24:
I do not use golang so by default GOPATH was not setup and emerging of the package was not successful. After setting it up and playing with symlinks I see the same error:

```
>>> Emerging (1 of 29) dev-embedded/arduino-builder-1.3.25::arduino-overlay
 * 1.3.25.tar.gz SHA256 SHA512 WHIRLPOOL size ;-) ...                                                                                                                                                                                 [ ok ]
>>> Unpacking source...
>>> Unpacking 1.3.25.tar.gz to /var/tmp/portage/dev-embedded/arduino-builder-1.3.25/work
>>> Source unpacked in /var/tmp/portage/dev-embedded/arduino-builder-1.3.25/work
>>> Preparing source in /var/tmp/portage/dev-embedded/arduino-builder-1.3.25/work/arduino-builder-1.3.25 ...
+ go fmt ./src/arduino.cc/...
+ go fix ./src/arduino.cc/...
+ go vet ./src/arduino.cc/...
src/arduino.cc/arduino-builder/main.go:43:2: cannot find package "arduino.cc/builder" in any of:
        /usr/lib/go/src/arduino.cc/builder (from $GOROOT)
        /usr/local/lib/go/src/arduino.cc/builder (from $GOPATH)
src/arduino.cc/arduino-builder/main.go:44:2: cannot find package "arduino.cc/builder/gohasissues" in any of:
        /usr/lib/go/src/arduino.cc/builder/gohasissues (from $GOROOT)
        /usr/local/lib/go/src/arduino.cc/builder/gohasissues (from $GOPATH)
src/arduino.cc/arduino-builder/main.go:45:2: cannot find package "arduino.cc/builder/i18n" in any of:
        /usr/lib/go/src/arduino.cc/builder/i18n (from $GOROOT)
        /usr/local/lib/go/src/arduino.cc/builder/i18n (from $GOPATH)
src/arduino.cc/arduino-builder/main.go:46:2: cannot find package "arduino.cc/builder/types" in any of:
        /usr/lib/go/src/arduino.cc/builder/types (from $GOROOT)
        /usr/local/lib/go/src/arduino.cc/builder/types (from $GOPATH)
src/arduino.cc/arduino-builder/main.go:47:2: cannot find package "arduino.cc/builder/utils" in any of:
        /usr/lib/go/src/arduino.cc/builder/utils (from $GOROOT)
        /usr/local/lib/go/src/arduino.cc/builder/utils (from $GOPATH)
src/arduino.cc/arduino-builder/main.go:48:2: cannot find package "arduino.cc/properties" in any of:
        /usr/lib/go/src/arduino.cc/properties (from $GOROOT)
        /usr/local/lib/go/src/arduino.cc/properties (from $GOPATH)
src/arduino.cc/arduino-builder/main.go:49:2: cannot find package "github.com/go-errors/errors" in any of:
        /usr/lib/go/src/github.com/go-errors/errors (from $GOROOT)
        /usr/local/lib/go/src/github.com/go-errors/errors (from $GOPATH)
src/arduino.cc/builder/container_find_includes.go:116:2: cannot find package "arduino.cc/builder/builder_utils" in any of:
        /usr/lib/go/src/arduino.cc/builder/builder_utils (from $GOROOT)
        /usr/local/lib/go/src/arduino.cc/builder/builder_utils (from $GOPATH)
src/arduino.cc/builder/add_additional_entries_to_context.go:35:2: cannot find package "arduino.cc/builder/constants" in any of:
        /usr/lib/go/src/arduino.cc/builder/constants (from $GOROOT)
        /usr/local/lib/go/src/arduino.cc/builder/constants (from $GOPATH)
src/arduino.cc/builder/ctags_runner.go:36:2: cannot find package "arduino.cc/builder/ctags" in any of:
        /usr/lib/go/src/arduino.cc/builder/ctags (from $GOROOT)
        /usr/local/lib/go/src/arduino.cc/builder/ctags (from $GOPATH)
src/arduino.cc/builder/builder.go:40:2: cannot find package "arduino.cc/builder/phases" in any of:
        /usr/lib/go/src/arduino.cc/builder/phases (from $GOROOT)
        /usr/local/lib/go/src/arduino.cc/builder/phases (from $GOPATH)
src/arduino.cc/builder/setup_build_properties.go:42:2: cannot find package "arduino.cc/timeutils" in any of:
        /usr/lib/go/src/arduino.cc/timeutils (from $GOROOT)
        /usr/local/lib/go/src/arduino.cc/timeutils (from $GOPATH)
src/arduino.cc/builder/test/helper.go:45:2: cannot find package "github.com/stretchr/testify/assert" in any of:
        /usr/lib/go/src/github.com/stretchr/testify/assert (from $GOROOT)
        /usr/local/lib/go/src/github.com/stretchr/testify/assert (from $GOPATH)
 * ERROR: dev-embedded/arduino-builder-1.3.25::arduino-overlay failed (prepare phase):
 *   (no error message)
 * 
 * Call stack:
 *     ebuild.sh, line 124:  Called src_prepare
 *   environment, line 1927:  Called die
 * The specific snippet of code:
 *       ./fmt_fix_vet || die;
 * 
 * If you need support, post the output of `emerge --info '=dev-embedded/arduino-builder-1.3.25::arduino-overlay'`,
 * the complete build log and the output of `emerge -pqv '=dev-embedded/arduino-builder-1.3.25::arduino-overlay'`.
 * The complete build log is located at '/var/tmp/portage/dev-embedded/arduino-builder-1.3.25/temp/build.log'.
 * The ebuild environment file is located at '/var/tmp/portage/dev-embedded/arduino-builder-1.3.25/temp/environment'.
 * Working directory: '/var/tmp/portage/dev-embedded/arduino-builder-1.3.25/work/arduino-builder-1.3.25'
 * S: '/var/tmp/portage/dev-embedded/arduino-builder-1.3.25/work/arduino-builder-1.3.25'

>>> Failed to emerge dev-embedded/arduino-builder-1.3.25, Log file:

```

All required dependencies now are on GOPATH:
```
ls -l /usr/local/lib/go/src/arduino.cc/
total 16
drwxr-xr-x  2 portage portage 4096 Mar 20  2017 arduino-builder
drwxr-xr-x 12 portage portage 4096 Mar 20  2017 builder
drwxr-xr-x  3 portage portage 4096 Mar 20  2017 properties
drwxr-xr-x  2 portage portage 4096 Mar 20  2017 timeutils

```
I do not understand why it does not see any dependencies even when it was installed manually.

mapmot, 2018-02-27:
Try FEATURES="-sandbox"

carbofos, 2018-02-27:
Nothing changed